EHammer98/DigitalPlastic #32: Accelerometer programmeren

OnurcanCekem, 2020-02-06:
Zie #7

De accelerometer Serial.out uitlezen en sturen naar de PC.
Herschrijf dit naar een functie.

EHammer98, 2020-02-06:
Vergeet niet om de [BOM-lijst](https://hogeschoolutrecht-my.sharepoint.com/:x:/r/personal/elwin_hammer_student_hu_nl/_layouts/15/Doc.aspx?sourcedoc=%7B7F750B4B-5406-427E-8965-1CF28875D625%7D&file=BOM_lijst.xlsx&action=default&mobileredirect=true) bij te werken als (eigen) hardware gaat gebruiken.

En als je iets gaat aansluiten aan de Arduino of bezig bent met firmware voor de Arduino, check altijd ff de [pinOut-lijst](https://hogeschoolutrecht-my.sharepoint.com/personal/elwin_hammer_student_hu_nl/_layouts/15/onedrive.aspx?FolderCTID=0x012000E942BD24E1EB8843A2C6E6F5DD408CA6&id=%2Fpersonal%2Felwin%5Fhammer%5Fstudent%5Fhu%5Fnl%2FDocuments%2FLeerjaar1%2FShared%2FDeltaRobot%5FPeriodeC%2FAfbeeldingen%2C%20Schemetics%20%26%20PinOuts%2FPinOut%2Etxt&parent=%2Fpersonal%2Felwin%5Fhammer%5Fstudent%5Fhu%5Fnl%2FDocuments%2FLeerjaar1%2FShared%2FDeltaRobot%5FPeriodeC%2FAfbeeldingen%2C%20Schemetics%20%26%20PinOuts) (en werk deze, in overleg bij).

OnurcanCekem, 2020-02-10:
Tot nu toe staat alles gereed qua raw data. Nu moet ik alleen nog berekenen hoe ik versnelling kan bereken. Welke formule pas ik toe op de raw data?
Vervolgens ik wil dit ook vragen aan de docent, want ik kom er zelf niet uit.

closed 2020-02-12 by OnurcanCekem

OnurcanCekem, 2020-02-17:
Alles zoals gevraagd werkt.
Ik kan nu de versnelling in g printen naar de serial monitor met de functie lees_accel_uit(int as);. 
Eventueel kan ik ook een raw value functie hiervan maken.